duff2013/Snooze #106: Error compiling on board Teensy 4.0 with RAW_HID

fbarbero, 2021-11-07:
Using Arduino 1.8.16 with Snooze v6.3.9. Here's my config:
[image]

I just need SnoozeDigital and really don't need USBSerial sleep functionality. It would be nice to have #defines to not include other drivers. Thanks!

Here's my simple code to test
```
#include <Snooze.h>
// Load drivers
SnoozeDigital digital;

int mode = 0;
// install drivers to a SnoozeBlock
SnoozeBlock config(digital);

void setup() 
{
Serial.begin(9600);
pinMode(LED_BUILTIN, OUTPUT);
Serial.println("Start");
digital.pinMode(6, INPUT_PULLUP, RISING);//pin, mode, type
digitalWrite(LED_BUILTIN, HIGH);
}

void loop() 
{
if (mode == 0)
{
  Snooze.deepSleep( config );// return module that woke processor
  mode = 1;
}
digitalWrite(LED_BUILTIN, HIGH);
delay(100);
digitalWrite(LED_BUILTIN, LOW);
delay(100);
digitalWrite(LED_BUILTIN, HIGH);
delay(200);
digitalWrite(LED_BUILTIN, LOW);
delay(500);
}
```

Here's the compiler error
```
"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -c -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=155 -DARDUINO=10816 -DARDUINO_TEENSY40 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IF:\\Users\\Ferdie\\AppData\\Local\\Temp\\arduino_build_669291/pch" "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4" "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\libraries\\Snooze\\src" "C:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\libraries\\Snooze\\src\\hal\\TEENSY_40\\SnoozeUSBSerial.cpp" -o "F:\\Users\\Ferdie\\AppData\\Local\\Temp\\arduino_build_669291\\libraries\\Snooze\\hal\\TEENSY_40\\SnoozeUSBSerial.cpp.o"
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src\hal\TEENSY_40\SnoozeDigital.cpp: In member function 'virtual void SnoozeDigital::enableDriver(uint8_t)':
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src\hal\TEENSY_40\SnoozeDigital.cpp:12:15: error: expected unqualified-id before numeric constant
 #define DR    0
               ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/imxrt.h:5664:33: note: in expansion of macro 'DR'
 #define GPIO6_DR   (IMXRT_GPIO6.DR)
                                 ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src\hal\TEENSY_40\SnoozeDigital.cpp:99:35: note: in expansion of macro 'GPIO6_DR'
                 case ( uint32_t )&GPIO6_DR: {
                                   ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src\hal\TEENSY_40\SnoozeDigital.cpp:12:15: error: expected ')' before numeric constant
 #define DR    0
               ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/imxrt.h:5664:33: note: in expansion of macro 'DR'
 #define GPIO6_DR   (IMXRT_GPIO6.DR)
                                 ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src\hal\TEENSY_40\SnoozeDigital.cpp:99:35: note: in expansion of macro 'GPIO6_DR'
                 case ( uint32_t )&GPIO6_DR: {
                                   ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src\hal\TEENSY_40\SnoozeDigital.cpp:502:1: error: expected ':' at end of input
 }
 ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src\hal\TEENSY_40\SnoozeDigital.cpp:502:1: error: expected statement at end of input
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src\hal\TEENSY_40\SnoozeDigital.cpp:502:1: error: expected '}' at end of input
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src\hal\TEENSY_40\SnoozeDigital.cpp:502:1: error: expected '}' at end of input
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src\hal\TEENSY_40\SnoozeDigital.cpp:502:1: error: expected '}' at end of input
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src\hal\TEENSY_40\SnoozeDigital.cpp:502:1: error: expected '}' at end of input
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src\hal\TEENSY_40\SnoozeCompare.cpp: In member function 'virtual void SnoozeCompare::disableDriver(uint8_t)':
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src\hal\TEENSY_40\SnoozeCompare.cpp:174:18: warning: 'IRQ_CMP' may be used uninitialized in this function [-Wmaybe-uninitialized]
     IRQ_NUMBER_t IRQ_CMP;
                  ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src\hal\TEENSY_40\SnoozeCompare.cpp: In member function 'virtual void SnoozeCompare::enableDriver(uint8_t)':
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze\src\hal\TEENSY_40\SnoozeCompare.cpp:302:18: warning: 'IRQ_CMP' may be used uninitialized in this function [-Wmaybe-uninitialized]
     IRQ_NUMBER_t IRQ_CMP;
                  ^
Using library Snooze at version 6.3.8 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Snooze 
Error compiling for board Teensy 4.0.
```

fbarbero, 2021-11-07:
I thought I could do similar [workaround as with Teensy LC issue](https://github.com/duff2013/Snooze/issues/105#issuecomment-962550712). I tried commenting out the SnoozeUSBSerial.h in hal.h and additionally replaced the MR and IMR defines to 

```
const int DR = 0;
const int IMR = 5;
```
Compiled SUCCESS! ------BUT, when ran, my board became unresponsive (hang).

gizatt, 2021-12-18:
+1. It appears to be a naming conflict with defines in the teensy library? Changing the defines to non-conflicting names or switching back to [teensy release 4.12.0 or ealier](https://github.com/platformio/platform-teensy/releases) resolves the compile issue.

However, I'm on a Teensy 4.1, and my board hangs too after fixing the issue. Limited probing on the `sleep_usb_serial` example makes it seem like the USB connection is never established, and if I skip the usb connection establishment, the board hangs somewhere in timer startup. But I don't know the library super very well and, since I'm not sure if t4.1 is even expected to work anyway, will hold off on probing farther!

mariosgit, 2024-05-31:
Just ran into this, without HID.
These defines are suboptimal.