duff2013/Snooze #108: Teensey 4.1 Support

c172jeff, 2021-12-23:
Hi,  I was wondering if this is working in Teensey 4.1?  The last time I checked, I had problems compiling.    I would like to have the Teensey 4.1 sleep at very low power and wake up ever 2 hours or so and perform like a bat out of hell/take pictures and save to the sd card.  What little I have read was related to clock scalers and 4.1 having lots of peripherals that needed to be switched off for low power..  For my Teensey 4.1 to be useful, it needs to sleep with extremenly low power current draw until needed...either time or interrupt
Jeff

savejeff, 2022-05-05:
It seems the library is no longer maintained. if you want to fix the code yourself its pretty easy. 

go to "Snooze/src/hal/TEENSY_40/SnoozeDigital.cpp" and fix the defines at the beginning to: 

```

#define _DR    0
#define _GDIR  1
#define _PSR   2
#define _ICR1  3
#define _ICR2  4
#define _IMR   5
#define _ISR   6
#define _EDGE  7

```

go through the code and add the underscore everywhere the defines are used. Its just a naming collision with register names that have changed I guess.  
compiles fine for me

c172jeff, 2022-05-06:
Thank you.  I will try that.

Jeff

dremeier, 2023-12-06:
I followed the suggestion and it works together with https://github.com/duff2013/Snooze/issues/114
Why the maintainer does not update the repro?

JHProjects, 2025-01-03:
Hello, I am currently trying to fix this, but changing those defines in this single file didn't solve the compiling error. I also did fix the other file from #114.  Any obvious idea what I might be doing wrong?

jaapiyo, 2025-06-10:
Indeed this seems to be broken to me as well. I'm getting errors about parameters that are not constant expressions. See the exerpt below:

```
In file included from C:\Users\jelstgeest\.platformio\packages\framework-arduinoteensy\cores\teensy4/core_pins.h:32,
                 from C:\Users\jelstgeest\.platformio\packages\framework-arduinoteensy\cores\teensy4/wiring.h:39,
                 from C:\Users\jelstgeest\.platformio\packages\framework-arduinoteensy\cores\teensy4/WProgram.h:46,
                 from C:\Users\jelstgeest\.platformio\packages\framework-arduinoteensy\cores\teensy4/Arduino.h:6,
                 from lib\Snooze\src\hal\TEENSY_40\SnoozeDigital.h:34,
                 from lib\Snooze\src\hal\TEENSY_40\SnoozeDigital.cpp:10:
lib\Snooze\src\hal\TEENSY_40\SnoozeDigital.cpp: In member function 'virtual void SnoozeDigital::enableDriver(uint8_t)':
C:\Users\jelstgeest\.platformio\packages\framework-arduinoteensy\cores\teensy4/imxrt.h:5782:35: error: 'reinterpret_cast<IMXRT_GPIO_t*>(1107296256)' is not a constant expression
 5782 | #define IMXRT_GPIO6             (*(IMXRT_GPIO_t *)IMXRT_GPIO6_ADDRESS)
      |                                   ^
C:\Users\jelstgeest\.platformio\packages\framework-arduinoteensy\cores\teensy4/imxrt.h:5783:42: note: in expansion of macro 'IMXRT_GPIO6'
 5783 | #define GPIO6_DR                        (IMXRT_GPIO6.DR)
      |                                          ^~~~~~~~~~~
lib\Snooze\src\hal\TEENSY_40\SnoozeDigital.cpp:99:35: note: in expansion of macro 'GPIO6_DR'
   99 |                 case ( uint32_t )&GPIO6_DR: {
```

DaveBKC, 2025-06-11:
> Indeed this seems to be broken to me as well. I'm getting errors about parameters that are not constant expressions. See the exerpt below:
>
> ```
> In file included from [...]

FYI - This project hasn't been updated in 5 years.