duff2013/Snooze #113: Snooze.h not working with Teensy 4 + Audio Shield

becauseinterwebs, 2022-03-14:
I have a sketch that I have converted to run on T4 + Audio shield from T3.2 + Audio shield.  Everything works fine, but the minute I include Snooze.h and load the sketch the T4 "bricks."   I tried running the examples provided, and same result.  I made just a simple sketch that prints out to the Serial monitor and again all I had to do was include Snooze.h and...brick.  I also made sure to change any digital pins that conflict with connected pins between the T4 and the AS.  I'm using the minimal pins needed to connect the T4 to the AS:

0 (GND), 6,7,8,10,11,12,13,15,18,19,20,21 and 3.3V

Not sure what else to try?  I noticed in the source that there is no Audio driver for the T4 like there is for the T3.2.

MenneBos, 2022-03-20:
I have a teensy 4.0 working with audio shield using snooze.h and digital wakeup. The code is not stable and stops after some cycles of sleep and I'm not sure it is caused by the sleep mode. To my knowledge deepsleep and hybernate do not work. I'm in the process of deleting functionalities, but after removing the sensors and interrupts the instability is still present. Thinking about moving back to a 3.2 as I need a battery.

closed 2022-04-13 by becauseinterwebs

BRAINDEX-PierreC, 2024-08-04:
Hello, is it possible to share the code?